sparkfun/SparkFun_VL53L1X_Arduino_Library #3: Implement ROI programming

artlyra, 2018-05-02:
According to the datasheet:
"ROI size and position are programmable: the user may chose a custom FoV from 4x4 SPADs (minimum size) up to 16x16 SPADs (full FoV). " This means the sensor could have a FOV angle as narrow as 15 deg.

chetansarmak, 2018-06-19:
Hey 
I wanted to create example for the VL53L1X singlemeasurement using API and involve ROI related functions also in it .Can you tell me how difficult is that to implement ? to run in raspberry pi

chetansarmak, 2018-06-25:
hey when I try to check the status of the error I get the status as following for VL53L1X for linux
Call of VL53L1_DataInit
API Status: -15 : 
API Status: -15 :

which is referred as "#define VL53L1_ERROR_DIVISION_BY_ZERO"
Any light on where am I going wrong?

Grahldg, 2018-09-07:
Hi @artlyra ,

I added a couple of methods for using custom ROI based on ST's API. They are the following:
```C
void VL53L1X::setUserRoi(UserRoi *roi)
```
and
```C
UserRoi* VL53L1X::getUserRoi()
```
As well as the following `struct`:
```C
typedef struct{
	uint8_t topLeftX;
	uint8_t topLeftY;
	uint8_t bottomRightX;
	uint8_t bottomRightY;
}UserRoi;
```
The _lower left_ corner of the array has coordinates (0, 0) and the _upper right_ has coordinates (15, 15) and you need to set the _upper left_ and _lower right_ which is somewhat counter-intuitive. I had to draw diagrams to make sure I was getting the correct coordinates.

If you have any more questions, please let me know.

closed 2018-10-15 by Grahldg