sparkfun/SparkFun_VL53L1X_Arduino_Library #5: Cablibration

Autobot86, 2018-05-14:
How to write the calibrations on this code,

PerformOffsetCalibration()
PerformXtalkCalibration()
GetCalibration data()

Grahldg, 2018-11-19:
Hi @Autobot86 ,

These are methods from the ST API that we have not yet implemented in our library.  Others have had success with the [Pololu VL53L1X port for Arduino](https://github.com/pololu/vl53l1x-st-api-arduino) which does include those methods for calibration.  Because their library was designed for a different product I cannot guarantee that it will work for you, but if you're feeling adventurous it could be a good reference for implementing them yourself.  If you do decide to go that route we would love it if you could send us a pull request with your changes.

You might also check out [our forums](https://forum.sparkfun.com/) to see if anyone else had success in implementing these methods.

Please let me know if you have any other questions.

AndyEngland521, 2019-01-02:
We've implemented a new version of the library that contains these functions @Autobot86

closed 2019-01-02 by AndyEngland521