sparkfun/SparkFun_VL53L1X_Arduino_Library #7: How to change I2C Address?

cpagravel, 2018-05-14:
I notice that the configBlock gets written to the chip every time you want to take a measurement (in the function startMeasurement). The configBlock's first byte is the device address. Is there a nice way to change the device's address then?

nseidle, 2018-05-14:
Great question. I noticed that too. I did a bit of testing and found that no, you cannot, as fair as I can tell, change the I2C address.

Also - you don't need to start the config block at 0x01 (I2C address register) every time. But we wanted to emulate the ST approach to eliminate as many unknowns as possible.

If anyone is able to figure out a way to change the I2C address, that would be great to know! I'll leave this open for now.

cpagravel, 2018-05-15:
I tried writing to the register and then changing the offset and it didn't seem to work. Also, the documentation says that the I2C address is 0x52 (http://www.st.com/content/ccc/resource/technical/document/datasheet/group3/7d/85/c8/95/fb/3b/4e/2d/DM00452094/files/DM00452094.pdf/jcr:content/translations/en.DM00452094.pdf).

In the code it uses 0x29 which is 1 bit shift away from 0x52. I'm not sure how this is working if that's the case. I found another repo (https://github.com/pololu/vl53l1x-st-api-arduino) that uses the 0x52 address and it works well (though I haven't tested it's ability to change I2C address yet).

So why is this one using 0x29? If it's a bug, then why is it working?

cpagravel, 2018-05-15:
Hmm... I could be misunderstanding how the I2C address is used. In the library I referenced above, the function for changing the device address shifts the input address by 2 before writing to the register. 

[image]

nseidle, 2018-05-15:
Within I2C the last bit of the first byte is the read/write bit. So the 'industry' gives addresses as 'unshifted 7-bit address'. I hate to say google it but it's fairly common at this point. Many/most datasheets for I2C devices will list the 7-bit address. 

Of course the ST datasheet makes me a liar ;) But when using Arduino (and ST's example code for that matter), you would use 

    Wire.beginTransmission(0x29) 

[Arduino docs](https://www.arduino.cc/en/Reference/WireBeginTransmission)

ralphhipps, 2018-08-29:
is there a way to use something similar to this to change the address on the ST sensor so you can have two or more on the same bus?

nseidle, 2018-08-29:
Currently it does not look like ST has built support for changing the sensor's I2C address. Because of this, the only way to use multiple sensors is to have two buses (such as the Teensy has two Wire ports) or use a multiplexer like our [Qwiic Mux Breakout](https://www.sparkfun.com/products/14685).

closed 2018-08-29 by nseidle

ralphhipps, 2018-08-29:
you may be correct, I'm having trouble with it. It seems you can write to
the register to change it, sure, but it breaks communication with the
sensor.

I'm not a sw whiz, so I can't tell you why.

A friend is using the VL53L0X and he can change the address no problem, but
perhaps the VL53L1X doesn't support that.

Ralph

>+<
 |    /\
 |___/  \
 [_@_]   `<
 O O O          DL Robotics!
------------------------------------------------------------------------------------
In the twenty-first century, the robot will take the place
which slave labor occupied in ancient civilization.
 -- Nikola Tesla
------------------------------------------------------------------------------------

cpagravel, 2018-08-29:
Sorry, I should've commented earlier.

I followed the instructions described here: https://www.pololu.com/file/0J1239/VL53L0X-AN4846.pdf

To summarize the document:
1. Put all the Vl53L1x chips on the same I2C bus
2. Connect each reset button to a separate I/O pin
3. Change the address of the sensors one at a time

ralphhipps, 2018-08-29:
> 0J1239/VL53L0X-AN4846.pdf

have you tried this?

I can't get it to work.

how do you do this on an Arduino with the given libs? it would seem you
need to have a lib from ST to make it work:

2.2 VL53L0X API management
See also “VL53L0X_API_Integration Guide.pdf” on www.st.com/VL53L0X in
“Design Resources” page. In vl53L0x_platform.h API file

• Set VL53L0x_SINGLE_DEVICE_DRIVER macro to 0 so that API implementation
will be automatically adapted to a multi-device context.
• Define VL53L0xDev_t type as a structure pointer holding any data required
for multidevice management. A mandatory field is an instance of
VL53L0xDevData containing ST API private data.
• Then
– define, “N” the number of VL53L0X (Struct MyVL53L0xDev_t BoardDevs[N])
– Put all devices under reset
– Enable VL53L0X one after the other and set their I2C address through
VL53L0X_SetDeviceAddress (&BoardDevs[i], FinalI2cAddr)

Ralph

>+<
 |    /\
 |___/  \
 [_@_]   `<
 O O O          DL Robotics!
------------------------------------------------------------------------------------
In the twenty-first century, the robot will take the place
which slave labor occupied in ancient civilization.
 -- Nikola Tesla
------------------------------------------------------------------------------------

cpagravel, 2018-08-29:
I have tried it and it works. I used the library from Pololu though. https://www.pololu.com/product/3415/resources

ralphhipps, 2018-08-29:
> It's a bit heavier, but it incorporates code written by ST.

you tried it on the VL53L0X or the VL53L1X?

Ralph

>+<
 |    /\
 |___/  \
 [_@_]   `<
 O O O          DL Robotics!
------------------------------------------------------------------------------------
In the twenty-first century, the robot will take the place
which slave labor occupied in ancient civilization.
 -- Nikola Tesla
------------------------------------------------------------------------------------

cpagravel, 2018-08-29:
VL53L1x

ralphhipps, 2018-08-29:
I'll try it again, thanks for confirming!

do you have an Arduino example .ino file I can use?

Ralph

>+<
 |    /\
 |___/  \
 [_@_]   `<
 O O O          DL Robotics!
------------------------------------------------------------------------------------
In the twenty-first century, the robot will take the place
which slave labor occupied in ancient civilization.
 -- Nikola Tesla
------------------------------------------------------------------------------------

cpagravel, 2018-08-29:
I don't sorry. It's in the company repo that I work for.

ralphhipps, 2018-08-29:
no worries, if I ever get this working I'll post something.

Ralph

>+<
 |    /\
 |___/  \
 [_@_]   `<
 O O O          DL Robotics!
------------------------------------------------------------------------------------
In the twenty-first century, the robot will take the place
which slave labor occupied in ancient civilization.
 -- Nikola Tesla
------------------------------------------------------------------------------------

MauroMombelli, 2018-09-07:
@cpagravel trying and NOT having success too, the sensor stop to answer. Do you change the address before or after the init()? is there any special stuff you do when you change it?

cpagravel, 2018-09-07:
I wish I could provide a snippet of code but I finished my internship and don't have access to the repo anymore. As a heads up, the VL53L1x sensors perform terribly in indirect sunlight. All I can say is that the order that you do things is important. It is documented, but you have to dig to find the answers--I recommend looking at the documentation linked to at the Pololu website or the ST website. I don't think this forum is an appropriate place to discuss the Pololu library. I'm sorry I can't be of more help.

killia66, 2018-10-09:
@ralphhipps did you bring it to work?

nseidle, 2018-11-30:
@wt70707 - Unfortunately ST did not allow for a direct way to modify the I2C address on the VL53L1X. The easiest way to attach multiple sensors is to use a [mux](https://www.sparkfun.com/products/14685).

killia66, 2018-11-30:
> 
> 
> Is it possible to change address with Sparkfun library?

you can change the address
there is a method to change the address but i don't know if the Sparfunk library implement this method

wt70707, 2018-11-30:
Thanks nseidle!
Thanks killia66, what is the method you mentioned to change the address? Is there any link for that? I googled and all for VL53L0X, not VL53L1X. I don't need to change the address online, so any method that can change the address is OK to have multiple VL53L1X on one I2C bus.

Shamshala, 2018-11-30:
> @wt70707 - Unfortunately ST did not allow for a direct way to modify the I2C address on the VL53L1X. The easiest way to attach multiple sensors is to use a [mux](https://www.sparkfun.com/products/14685).

Than why/how is it possible via pololu library? Btw: The interesting thing is that VL53L1X remembers the changed address (VL53L0X resets address after power suppply cutoff)

Try it on your own:
`#include <Wire.h>
#include <VL53L1X.h>

VL53L1X sensor;
VL53L1X sensor2;

void setup()
{

  pinMode(16, OUTPUT);
  pinMode(3, OUTPUT);
  digitalWrite(16, LOW);  
  digitalWrite(3, LOW);

  delay(500);
  Wire.begin();

  Serial.begin (115200);

  pinMode(3, INPUT);
  delay(150);
  Serial.println("00");
  sensor.init(true);

  Serial.println("01");
  delay(100);
  sensor.setAddress((uint8_t)22);
  Serial.println("02");

  pinMode(16, INPUT);
  delay(150);
  sensor2.init(true);
  Serial.println("03");
  delay(100);
  sensor2.setAddress((uint8_t)43);
  Serial.println("04");

  Serial.println("addresses set");

Serial.println ("I2C scanner. Scanning ...");
  byte count = 0;

  for (byte i = 1; i < 120; i++)
  {

    Wire.beginTransmission (i);
    if (Wire.endTransmission () == 0)
    {
      Serial.print ("Found address: ");
      Serial.print (i, DEC);
      Serial.print (" (0x");
      Serial.print (i, HEX);
      Serial.println (")");
      count++;
      delay (1); 
    }
  }
  Serial.println ("Done.");
  Serial.print ("Found ");
  Serial.print (count, DEC);
  Serial.println (" device(s).");
  
  sensor.setTimeout(500);
  sensor2.setTimeout(500);

  sensor.startContinuous();
  sensor2.startContinuous();

}

void loop()
{
  Serial.print(sensor.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor2.readRangeContinuousMillimeters());
  Serial.println();
}`

Shamshala, 2018-11-30:
> @wt70707 - Unfortunately ST did not allow for a direct way to modify the I2C address on the VL53L1X. The easiest way to attach multiple sensors is to use a [mux](https://www.sparkfun.com/products/14685).

Than why/how is it possible via pololu library? Btw: The interesting thing is the VL53L1X remembers the changed address (VL53L0X resets address after power suppply cutoff)

Try it on your own:
```
#include <Wire.h>
#include <VL53L1X.h>

VL53L1X sensor;
VL53L1X sensor2;

void setup()
{

  pinMode(16, OUTPUT);
  pinMode(3, OUTPUT);
  digitalWrite(16, LOW);  
  digitalWrite(3, LOW);

  delay(500);
  Wire.begin();

  Serial.begin (115200);

  pinMode(3, INPUT);
  delay(150);
  Serial.println("00");
  sensor.init(true);

  Serial.println("01");
  delay(100);
  sensor.setAddress((uint8_t)22);
  Serial.println("02");

  pinMode(16, INPUT);
  delay(150);
  sensor2.init(true);
  Serial.println("03");
  delay(100);
  sensor2.setAddress((uint8_t)43);
  Serial.println("04");

  Serial.println("addresses set");

Serial.println ("I2C scanner. Scanning ...");
  byte count = 0;

  for (byte i = 1; i < 120; i++)
  {

    Wire.beginTransmission (i);
    if (Wire.endTransmission () == 0)
    {
      Serial.print ("Found address: ");
      Serial.print (i, DEC);
      Serial.print (" (0x");
      Serial.print (i, HEX);
      Serial.println (")");
      count++;
      delay (1); 
    }
  }
  Serial.println ("Done.");
  Serial.print ("Found ");
  Serial.print (count, DEC);
  Serial.println (" device(s).");
  
  sensor.setTimeout(500);
  sensor2.setTimeout(500);

  sensor.startContinuous();
  sensor2.startContinuous();

}

void loop()
{
  Serial.print(sensor.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor2.readRangeContinuousMillimeters());
  Serial.println();
}

Shamshala, 2018-11-30:
> @wt70707 - Unfortunately ST did not allow for a direct way to modify the I2C address on the VL53L1X. The easiest way to attach multiple sensors is to use a [mux](https://www.sparkfun.com/products/14685).

Then why/how is it possible via pololu library?

Try it on your own (code for two sensors, one I2C):

```
#include <Wire.h>
#include <VL53L1X.h>

VL53L1X sensor;
VL53L1X sensor2;

void setup()
{

  pinMode(16, OUTPUT);
  pinMode(3, OUTPUT);
  digitalWrite(16, LOW);  
  digitalWrite(3, LOW);

  delay(500);
  Wire.begin();

  Serial.begin (115200);

  pinMode(3, INPUT);
  delay(150);
  Serial.println("00");
  sensor.init(true);

  Serial.println("01");
  delay(100);
  sensor.setAddress((uint8_t)22);
  Serial.println("02");

  pinMode(16, INPUT);
  delay(150);
  sensor2.init(true);
  Serial.println("03");
  delay(100);
  sensor2.setAddress((uint8_t)43);
  Serial.println("04");

  Serial.println("addresses set");

Serial.println ("I2C scanner. Scanning ...");
  byte count = 0;

  for (byte i = 1; i < 120; i++)
  {

    Wire.beginTransmission (i);
    if (Wire.endTransmission () == 0)
    {
      Serial.print ("Found address: ");
      Serial.print (i, DEC);
      Serial.print (" (0x");
      Serial.print (i, HEX);
      Serial.println (")");
      count++;
      delay (1); 
    }
  }
  Serial.println ("Done.");
  Serial.print ("Found ");
  Serial.print (count, DEC);
  Serial.println (" device(s).");
  
  sensor.setTimeout(500);
  sensor2.setTimeout(500);

  sensor.startContinuous();
  sensor2.startContinuous();

}

void loop()
{
  Serial.print(sensor.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor2.readRangeContinuousMillimeters());
  Serial.println();
}

```
Btw, interesting thing: The VL53L1X remembers the changed address. VL53L0X resets address after power suppply cutoff.

wt70707, 2018-11-30:
@Shamshala Thanks. Will your code work with one VL53L1X? So your code works, so we can change address online in realtime, right?

killia66, 2018-12-01:
> 
> 
> Thanks nseidle!
> Thanks killia66, what is the method you mentioned to change the address? Is there any link for that? I googled and all for VL53L0X, not VL53L1X. I don't need to change the address online, so any method that can change the address is OK to have multiple VL53L1X on one I2C bus.

i tried it with a related code like Shamshala post it. it worked but my sensors got different address then i set to them
ST said to change the address it is the same way as with the sensor VL53L0X
https://www.st.com/content/ccc/resource/technical/document/application_note/group0/0e/0a/96/1b/82/19/4f/c2/DM00280486/files/DM00280486.pdf/jcr:content/translations/en.DM00280486.pdf

wt70707, 2018-12-01:
@Shamshala I tried your code on one and two VL53L1X (SparkFun and Penky on both Uno and Teensy 3.6. Sometime, the address can be changed, but after power cycle, the address is still 0x29. And when run the code, SDL and SDA drop to 1.7V (normally 3.3V). Why set up pin 16 and pin 3? What is the purpose to operate the two pins? Should I put VL53L1X on special pins?

MauroMombelli, 2018-12-01:
@wt70707 the documentation is clear, you can change address but it NOT persistent, and you have to set it up every reboot.

This is my solution for 16 sensor: https://github.com/5N44P/53xLIDAR

Shamshala, 2018-12-02:
> @Shamshala Thanks. Will your code work with one VL53L1X? So your code works, so we can change address online in realtime, right?

I'm not quite sure what you mean by "online in realtime". If you think to change address multiple times in a loop, i haven't tried such thing.

> i tried it with a related code like @Shamshala post it. it worked but my sensors got different address then i set to them
> ST said to change the address it is the same way as with the sensor VL53L0X [...]

That's weird, never saw this behaviour.

> @Shamshala I tried your code on one and two VL53L1X (SparkFun and Penky on both Uno and Teensy 3.6. Sometime, the address can be changed, but after power cycle, the address is still 0x29. And when [...]

Of course you can change those pins, i've been testing this code in NodeMCU ESP8266 and these pins suited to my use at that time. As well as you can delete about half of the code(checking addresses and printing info about it). And if you need 10-20 sensors, sure thing:
#include <Wire.h>
#include <VL53L1X.h>

VL53L1X sensor;
VL53L1X sensor2;
VL53L1X sensor3;
VL53L1X sensor4;
VL53L1X sensor5;
VL53L1X sensor6;
VL53L1X sensor7;
VL53L1X sensor8;
VL53L1X sensor9;
VL53L1X sensor10;

void setup()
{

  pinMode(1, OUTPUT);
  pinMode(2, OUTPUT);
  pinMode(3, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(6, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(8, OUTPUT);
  pinMode(9, OUTPUT);
  pinMode(10, OUTPUT);
  digitalWrite(1, LOW);  
  digitalWrite(2, LOW);
  digitalWrite(3, LOW);  
  digitalWrite(4, LOW);
  digitalWrite(5, LOW);  
  digitalWrite(6, LOW);
  digitalWrite(7, LOW);  
  digitalWrite(8, LOW);
  digitalWrite(9, LOW);  
  digitalWrite(10, LOW);

  delay(500);
  Wire.begin();

  Serial.begin (115200);

  pinMode(3, INPUT);
  delay(150);
  sensor.init(true);
  delay(100);
  sensor.setAddress((uint8_t)22);
  Serial.println("02");

  pinMode(16, INPUT);
  delay(150);
  sensor2.init(true);
  delay(100);
  sensor2.setAddress((uint8_t)43);

  sensor.setTimeout(500);
  sensor2.setTimeout(500);
  sensor.setTimeout(500);
  sensor2.setTimeout(500);
  sensor.setTimeout(500);
  sensor2.setTimeout(500);
  sensor.setTimeout(500);
  sensor2.setTimeout(500);

  sensor.startContinuous();
  sensor2.startContinuous();
  sensor3.startContinuous();
  sensor4.startContinuous();
  sensor5.startContinuous();
  sensor6.startContinuous();
  sensor7.startContinuous();
  sensor8.startContinuous();
  sensor9.startContinuous();
  sensor10.startContinuous();

}

void loop()
{
  Serial.print(sensor.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor2.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor3.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor4.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor5.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor6.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor7.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor8.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor9.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor10.readRangeContinuousMillimeters());
  Serial.println();
}

Shamshala, 2018-12-02:
> @wt70707 Thanks. Will your code work with one VL53L1X? So your code works, so we can change address online in realtime, right?

Ok, first of all, it's not my code. I've picked it up somewhere when i've been testing multiple VL53L0Xs. And because the VL53L1X communication looked pretty similar to the old one, i've just gave it a shot when i've noticed Pololu has support for address change (if i remember correctly, i just had to edit some minor stuff).
I'm not quite sure what you mean by "online in realtime". If you think to change address multiple times in a loop, i haven't tried such thing.

> @killia66  i tried it with a related code like @Shamshala post it. it worked but my sensors got different address then i set to them
> ST said to change the address it is the same way as with the sensor VL53L0X
> https://www.st.com/content/ccc/resource/technical/document/application_note/group0/0e/0a/96/1b/82/19/4f/c2/DM00280486/files/DM00280486.pdf/jcr:content/translations/en.DM00280486.pdf

Have you connected the XSHUT pin?

> @wt70707  I tried your code on one and two VL53L1X (SparkFun and Penky on both Uno and Teensy 3.6. Sometime, the address can be changed, but after power cycle, the address is still 0x29. And when run the code, SDL and SDA drop to 1.7V (normally 3.3V). Why set up pin 16 and pin 3? What is the purpose to operate the two pins? Should I put VL53L1X on special pins?

Of course you can change those pins. I've been testing this code in NodeMCU ESP8266 and these pins suited to my use at that time. As well as you can delete about half of the code(checking addresses and printing info about it). And if you need 10-20 sensors, i THINK there are "no limitations", so sure thing: (it should work but i edit it on my phone, so excuse any mistakes - but it's prettystraight forward)
```
#include <Wire.h>
#include <VL53L1X.h>

VL53L1X sensor0;
VL53L1X sensor1;
VL53L1X sensor2;
VL53L1X sensor3;
VL53L1X sensor4;
VL53L1X sensor5;
VL53L1X sensor6;
VL53L1X sensor7;
VL53L1X sensor8;
VL53L1X sensor9;

void setup()
{

  pinMode(1, OUTPUT);
  pinMode(2, OUTPUT);
  pinMode(3, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(6, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(8, OUTPUT);
  pinMode(9, OUTPUT);
  pinMode(10, OUTPUT);
  digitalWrite(1, LOW);  
  digitalWrite(2, LOW);
  digitalWrite(3, LOW);  
  digitalWrite(4, LOW);
  digitalWrite(5, LOW);  
  digitalWrite(6, LOW);
  digitalWrite(7, LOW);  
  digitalWrite(8, LOW);
  digitalWrite(9, LOW);  
  digitalWrite(10, LOW);

  delay(500);
  Wire.begin();

  Serial.begin (115200);

  pinMode(10, INPUT);
  delay(150);
  sensor0.init(true);
  delay(100);
  sensor0.setAddress((uint8_t)22);

  pinMode(1, INPUT);
  delay(150);
  sensor1.init(true);
  delay(100);
  sensor1.setAddress((uint8_t)23);

  pinMode(2, INPUT);
  delay(150);
  sensor2.init(true);
  delay(100);
  sensor2.setAddress((uint8_t)24);

  pinMode(3, INPUT);
  delay(150);
  sensor3.init(true);
  delay(100);
  sensor3.setAddress((uint8_t)25);

  pinMode(4, INPUT);
  delay(150);
  sensor4.init(true);
  delay(100);
  sensor4.setAddress((uint8_t)26);

  pinMode(5, INPUT);
  delay(150);
  sensor5.init(true);
  delay(100);
  sensor5.setAddress((uint8_t)27);

  pinMode(6, INPUT);
  delay(150);
  sensor6.init(true);
  delay(100);
  sensor6.setAddress((uint8_t)28);

  pinMode(7, INPUT);
  delay(150);
  sensor7.init(true);
  delay(100);
  sensor7.setAddress((uint8_t)29);

  pinMode(8, INPUT);
  delay(150);
  sensor8.init(true);
  delay(100);
  sensor8.setAddress((uint8_t)30);

  pinMode(9, INPUT);
  delay(150);
  sensor9.init(true);
  delay(100);
  sensor9.setAddress((uint8_t)31);

  sensor0.setTimeout(500);
  sensor1.setTimeout(500);
  sensor2.setTimeout(500);
  sensor3.setTimeout(500);
  sensor4.setTimeout(500);
  sensor5.setTimeout(500);
  sensor6.setTimeout(500);
  sensor7.setTimeout(500);
  sensor8.setTimeout(500);
  sensor9.setTimeout(500);

  sensor0.startContinuous();
  sensor1.startContinuous();
  sensor2.startContinuous();
  sensor3.startContinuous();
  sensor4.startContinuous();
  sensor5.startContinuous();
  sensor6.startContinuous();
  sensor7.startContinuous();
  sensor8.startContinuous();
  sensor9.startContinuous();

}

void loop()
{
  Serial.print(sensor0.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor1.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor2.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor3.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor4.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor5.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor6.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor7.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor8.readRangeContinuousMillimeters());
  Serial.print(',');
  Serial.print(sensor9.readRangeContinuousMillimeters());
  Serial.println();
}
```
MauroMombelli is right about (non)persistance of the address, so you have to set the addresses on every powercycle.
But as i've said before, i don't know why, my VL53L1X remembers the address ;-) (it took me about a minute when i switched back to Sparkfun library and i didn't get any output...

MauroMombelli, 2018-12-02:
I used VL53L1X and then dont.. Are you dure about the code you are running
doea not set then up at the beginnig?
Also, I had issue with some sensor symply not working at all (no answer
from i2c), bit i guess that is another kind of problemi, just wondering if
that happen to you too.

wt70707, 2018-12-02:
@MauroMombelli Thanks! As you said ST doesn't design to have the I2C address "permanently" changed, I guess you code also should do the change in the code every time?

wt70707, 2018-12-02:
@Shamshala, great thanks. Now I understand your pins operations: I need to connect the Shut pin to one of the GPIOs to make the change of address effect. I will try that.

wt70707, 2018-12-02:
@Shamshala. I got the strange that after running your code (I didn't connect the SHUT yet), SDL and SDA drops to 1.7V. I use Teensy 3.6.

MauroMombelli, 2018-12-02:
> I guess you code also should do the change in the code every time?

exactly

wt70707, 2018-12-03:
@Shamshala, I got confirmed response from ST, the changed address cannot be memorized. I connect the SHUT to one GPIO, and your code works fine with one or multiple VL53L1X. I tried both Sparkfun version and Pesky version.

eddie0x7c3, 2019-03-05:
I am not sure if you guys figured this out but to change the adress you call that function form the api 
VL53L0X_SetDeviceAddress
then you must update the address in your structure.
and anytime you lose power on the xshut pin the address resets to defualt so you have to handle that on the power up of xshut pin. of course I am using the API straight up with no arduino or anything so im not sure if you guys know what structure im talking about. but i have been scratching my head as to why it wasnt working and its just that during power cycles of the xshut pin it resets.

MauroMombelli, 2019-03-05:
This api is different from the official one and does not have that
functionality.

jendy12345, 2023-06-12:
Hello, I have purchased the product with part number 100080101-01-EVAL, which is listed as part number 2190-100080101-01-EVAL-ND. I encountered an issue where I need to change the sensor address. For example, I want to change the original sensor address from 18 to 30. Initially, I was able to successfully update the address, but after connecting via Bluetooth and powering off, the previously set sensor address reverted back to the default value and I am unable to change the address anymore. What could be the cause of this issue? How should I proceed if I still need to change the address?

I have encountered the same issue with both the Bluetooth module on the Bluno Beetle v1.1 and the HC-06 Bluetooth module connected to the Arduino UNO board. After modifying the address, it reverts back to the default address and cannot be changed again. Does this sensor support address modification? Currently, I have two devices with the address set to 18, but I need to modify three different addresses: 18, 30, and 40. How should I proceed to handle this?

I'm using this code to change the address. However, when I try to modify it, it doesn't change successfully in one attempt. I need to repeatedly refresh and modify it to make the change successful.I used the Example6_SetI2CAddress code file provided to make the modification.

[image]

Another issue I encountered is that I can successfully change the address when I set it to 18, but any other address setting fails.
It is possible that the sensor's address modification is locked or restricted？？

[image]