sparkfun/SparkFun_VL53L1X_Arduino_Library #9: Taking advantage of the VL53L1_set_user_zone function

dankelley2, 2018-06-07:
Forgive me for my novice understanding(I am fairly new to C++ and C), but is there a way to take advantage of this convenience function?

vl53l1_api_core.c Line 1050:
[image]

I understand this sensor has a roughly 27 degree FOV, does anybody know if that is while using the smallest SPAD array size possible? (4x4 I believe)

nseidle, 2018-06-08:
I believe this is a duplicate of [this issue](https://github.com/sparkfun/SparkFun_VL53L1X_Arduino_Library/issues/3). Let me know if I'm mistaken.

We don't implement the full ST driver so can can't use this function easily.  But we are attempting to implement SPAD array configuration. It's *hard* and has a lot of unintended consequences that we're working with ST on.

ash73, 2018-08-04:
Any progress on this? I'm hoping to use a VL53L1X to measure distances inside a narrow (140mm) tube which would require the minimum field of view (15 deg) to measure only 50cm. The pololu ST VL53L1X API implementation has the SetUserROI function but doesn't seem to work on a ESP32 Arduino, whereas this SparkFun library does work but doesn't have the ROI function. Just my luck!

Grahldg, 2018-09-07:
Hello @dankelley2  and @ash73 ,

I added a couple of methods for using custom ROI based on  ST's API.  They are the following:
```C
void VL53L1X::setUserRoi(UserRoi *roi)
```
and
```C
UserRoi* VL53L1X::getUserRoi()
```

As well as the following `struct`:
```C
typedef struct{
	uint8_t topLeftX;
	uint8_t topLeftY;
	uint8_t bottomRightX;
	uint8_t bottomRightY;
}UserRoi;
```

The _lower left_ corner of the array has coordinates (0, 0) and the _upper right_ has coordinates (15, 15) and you need to set the _upper left_ and _lower right_ which is somewhat counter-intuitive.  I had to draw diagrams to make sure I was getting the correct coordinates.

If you have any more questions, please let me know.

LGKev, 2018-09-12:
@dankelley2 

> I understand this sensor has a roughly 27 degree FOV, does anybody know if that is while using the smallest SPAD array size possible? (4x4 I believe)

Section 3.6 of the Data sheet lists the Field of View for varying array sizes. 27 degree for a 16x16 array. 4x4 has a Field of View of 15 degrees.