sparkfun/SparkFun_VL53L1X_Arduino_Library #15: ld error

RichardZeitler, 2018-10-27:
Example1_ReadDistance.cpp.o: In Funktion »_GLOBAL__sub_I_distanceSensor«:
/home/varg/sketchbook/libraries/SparkFunSensor/src/SparkFun_VL53L1X_Arduino_Library.h:90: Warnung: undefinierter Verweis auf »VL53L1X::begin(unsigned char, TwoWire&)«
/home/varg/sketchbook/libraries/SparkFunSensor/src/SparkFun_VL53L1X_Arduino_Library.h:90: Warnung: undefinierter Verweis auf »VL53L1X::newDataReady()«
/home/varg/sketchbook/libraries/SparkFunSensor/src/SparkFun_VL53L1X_Arduino_Library.h:90: Warnung: undefinierter Verweis auf »VL53L1X::getDistance()«
collect2: Fehler: ld gab 1 als Ende-Status zurück

nseidle, 2018-11-03:
Hi, please give us some additional information:

### Subject of the issue
Describe your issue here.

### Your workbench
* What platform are you using?
* What version of the device are you using? Is there a firmware version?
* How is the device wired to your platform?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

AndyEngland521, 2019-12-30:
Closing due to inactivity. Feel free to reopen if issue persists.

closed 2019-12-30 by AndyEngland521