sparkfun/SparkFun_VL53L1X_Arduino_Library #31: Measure 2 distances, same sensor

elpedriyo, 2019-10-28:
Hello,

I have checked this video: https://www.youtube.com/watch?v=c91Ve-g0J2U for the sensor vl53l1x that was created by STMicroelectronics and it seems they use some kind of the same sensor to take 2 different readings I guess, do you have any clue on how could I achieve this with your library, or how are they doing it?

Kind regards

Muny, 2019-11-02:
The way they're doing this is by changing the center of a rectangular ROI (Region of Interest) on-the-fly. 

Setting the center of the ROI is currently not possible with the SparkFun or Pololu libraries unfortunately.

You can find the source for the demo in that video here: https://www.st.com/en/embedded-software/stsw-img010.html

In 'Src/main.c', line 282:
```
...
status = VL53l1X_SetROICenter(dev, center[Zone]);
...
```

AndyEngland521, 2019-11-04:
Hello @elpedriyo and @Muny 
I've added support for setting the optical center in v1.2.0

closed 2019-11-04 by AndyEngland521