sparkfun/SparkFun_VL53L1X_Arduino_Library #32: 1V8 vs 2V8 Mode

Muny, 2019-11-02:
The VL53L1X datasheet says there are two modes which correspond to different IOVDD voltages:

Section 5.2
https://www.st.com/resource/en/datasheet/vl53l1x.pdf

The VL53L1X user manual then tells the user how to switch between these modes (this seems to be for the old API):

Section 2.5.9
https://www.st.com/content/ccc/resource/technical/document/user_manual/group0/98/0d/38/38/5d/84/49/1f/DM00474730/files/DM00474730.pdf/jcr:content/translations/en.DM00474730.pdf

In the register configuration array in the code here, there seems to be a contradiction:
https://github.com/sparkfun/SparkFun_VL53L1X_Arduino_Library/blob/1773c55cb9fe2427d5f6ceef2aaf080b380fc88f/src/vl53l1x_class.cpp#L54-L55

Unless I'm misunderstanding this, it seems that bit 0 should be set for both of these register values. The two pins ~INT and ~SHUT are pulled up to AVDD (3.3V) on the SparkFun board.

Does anyone know what these registers actually do? Do they enable some internal pull-ups which are not used here because the SparkFun board has pull-ups?

The pololu library does seem to support switching between the modes:
https://github.com/pololu/vl53l1x-arduino/blob/0636d160b61c54ee4ae249427fa8d67efb818f06/VL53L1X.cpp#L30-L35

Their board also pulls-up these pins to AVDD.

santaimpersonator, 2019-12-28:
I'd have to double check with the engineer on the values and comments for those configuration settings; I think they were provided to us by ST. Unfortunately, for more information on what those registers actually do, you'll probably have to contact ST.

Currently, we have not heard of any issues regarding this so far.

AndyEngland521, 2019-12-30:
@Muny I believe you're correct in that bit 1 should be set in these and setting it in the library seems to have no effect on functionality. Like you, I'm unsure what exactly these registers do and have emailed ST to get a better handle on what's happening under the hood. Although your suggested settings work, I'm going to wait to change the library until we know exactly what's going on.

Nice catch by the way!

AndyEngland521, 2020-01-06:
Hey @Muny here's a reply from my colleague at ST in regards to these settings

"The I2C ‘clocks’ a bit when the SCL drops. The setting effects when the transition is made.
In theory the best place to note the transition is at ½ the VDD.
But the sensor doesn’t ‘know’ where that is.
So the register simply tells the sensor when to clock.

I’m not totally sure it matters.
Clearly it’s an I2C register setting and you cannot set it unless the I2C is working.
But one half of 1v8 is 0.9 and one half of 2v8 is 1.4. So we are worried about a half volt.
And a good design will drive from 1V8 to 0, so when the clock triggers is arbitrary.

But I2C is really subject to noise, and a minor glitch just at the transition point will cause and extra clock and screw everything up."

It seems that it's probably a decent idea to set these for noisier applications

AndyEngland521, 2020-01-06:
@Muny fixed in v1.2.2

closed 2020-01-06 by AndyEngland521

Muny, 2020-01-06:
Great to hear. Your effort and your colleague's explanation are very much appreciated :)