sparkfun/SparkFun_VL53L1X_Arduino_Library #40: Implement setDistanceModeMedium()

jimtng, 2020-08-17:
What does it take to implement setDistanceModeMedium() please? Thanks!

AndyEngland521, 2020-08-17:
Interesting question with a slight problem, we don't have a register map for the VL53L1X. ST was kind enough to provide us with a lightweight library, check out how they implement `setDistanceMode` below.
```
VL53L1X_ERROR VL53L1X::VL53L1X_SetDistanceMode(uint16_t DM)
{
	uint16_t TB;
	VL53L1X_ERROR status = 0;

	status = VL53L1X_GetTimingBudgetInMs(&TB);
	switch (DM)
	{
	case 1:
		status = VL53L1_WrByte(Device, PHASECAL_CONFIG__TIMEOUT_MACROP, 0x14);
		status = VL53L1_WrByte(Device, RANGE_CONFIG__VCSEL_PERIOD_A, 0x07);
		status = VL53L1_WrByte(Device, RANGE_CONFIG__VCSEL_PERIOD_B, 0x05);
		status = VL53L1_WrByte(Device, RANGE_CONFIG__VALID_PHASE_HIGH, 0x38);
		status = VL53L1_WrWord(Device, SD_CONFIG__WOI_SD0, 0x0705);
		status = VL53L1_WrWord(Device, SD_CONFIG__INITIAL_PHASE_SD0, 0x0606);
		break;
	case 2:
		status = VL53L1_WrByte(Device, PHASECAL_CONFIG__TIMEOUT_MACROP, 0x0A);
		status = VL53L1_WrByte(Device, RANGE_CONFIG__VCSEL_PERIOD_A, 0x0F);
		status = VL53L1_WrByte(Device, RANGE_CONFIG__VCSEL_PERIOD_B, 0x0D);
		status = VL53L1_WrByte(Device, RANGE_CONFIG__VALID_PHASE_HIGH, 0xB8);
		status = VL53L1_WrWord(Device, SD_CONFIG__WOI_SD0, 0x0F0D);
		status = VL53L1_WrWord(Device, SD_CONFIG__INITIAL_PHASE_SD0, 0x0E0E);
		break;
	default:
		break;
	}
	status = VL53L1X_SetTimingBudgetInMs(TB);
	return status;
}
```

In the above block, case 2 is long distance while 1 is short, we can see which registers are being written to, but don't necessarily know what the values being written represent. Getting the proper values for mid range would be guesswork without a dev kit and some I2C lines to sniff, so let me contact some folks with access to the register map and see if I can drum up a solution.

jimtng, 2020-08-17:
Thanks for looking into this. I was going to do a PR but I haven't got all the pieces of the puzzle, so instead, I'd just post here what I've found so far.

At a guess based on scouring some files I can find out there, this could be the needed values:

```
	case 3: // medium
		status += VL53L1_WrByte(Device, PHASECAL_CONFIG__TIMEOUT_MACROP, 0x0D); // very uncertain about this particular one
		status += VL53L1_WrByte(Device, RANGE_CONFIG__VCSEL_PERIOD_A, 0x0B);
		status += VL53L1_WrByte(Device, RANGE_CONFIG__VCSEL_PERIOD_B, 0x09);
		status += VL53L1_WrByte(Device, RANGE_CONFIG__VALID_PHASE_HIGH, 0x78);
		status += VL53L1_WrWord(Device, SD_CONFIG__WOI_SD0, 0x0B09);
		status += VL53L1_WrWord(Device, SD_CONFIG__INITIAL_PHASE_SD0, 0x0A0A);
		break;
```

Note that in ST's original code they kept adding the `status += ` although that's irrelevant here.

What I haven't managed to deduce yet is what should go inside `VL53L1X::VL53L1X_SetTimingBudgetInMs` for the medium   distance mode (DM=3 above)

* VL53L1X::VL53L1X_GetTimingBudgetInMs will also need to be updated
* VL53L1X::VL53L1X_GetDistanceMode
```
VL53L1X_ERROR VL53L1X::VL53L1X_GetDistanceMode(uint16_t *DM)
{
	uint8_t TempDM, status = 0;

	status = VL53L1_RdByte(Device, PHASECAL_CONFIG__TIMEOUT_MACROP, &TempDM);
	switch(TempDM) {
		case 0x14: *DM = 1; break;
		case 0x0A: *DM = 2; break;
		case 0x0D: *DM = 3; break;
	}
	return status;
}
```

There may be other spots I've missed too.

AndyEngland521, 2020-08-17:
Great work on the deduction! Awaiting word from ST, they're usually pretty good about helping us so stay tuned!

AndyEngland521, 2020-08-20:
Wow @jimtng I just got an email back from ST and it seems that you've nailed it!! if you have everything organized into a PR, go ahead and bump the version number to 1.2.10 in the library.properties and file. Otherwise I can implement your changes and go ahead and push.

geologic, 2022-06-15:
Any news about this issue? I'm also interested on medium distance mode...

jimtng, 2023-10-11:
I've switched to Pololu's library which supports Medium distance mode and seems to be more polished.

closed 2023-10-11 by jimtng