sparkfun/SparkFun_VL53L1X_Arduino_Library #43: Synchronise getDistance and getRangeStatus

hb020, 2020-12-26:
getDistance and getRangeStatus go together. No use reading getDistance if one does not know the range status, as the status will tell if the distance read is valid or not.
However, in the SDK they are 2 separate calls.
When in continuous mode, therefore, it is not always clear if the values that are read pertain to the same sampling.
This can be tested easily: 
```
s1 = ds->getRangeStatus();
d = ds->getDistance();
s2 = ds->getRangeStatus();
if (s1 != s2) { .. signal error or ignore... }
```
When continuous mode is enabled, over time, you will get this error, especially if the reading conditions are bad.
 
It is best to read the related registers all at the same time from the device, as can be found in some of the other VL53L1X libs.

jimtng, 2022-02-01:
@hb020 could you point me to the other libs which implemented reading the related registers at the same time?

hb020, 2022-02-01:
In the end I went with the "VL53L1X" pololu lib. It has less functionality as far as I remember, but does read correctly (see VL53L1X::readResults).  I guess some others also can do the job.