sparkfun/SparkFun_VL53L1X_Arduino_Library #46: Unable to add library with Arduino IDE 2.0.0-beta.7

ernestp, 2021-06-24:
Error: Request installZip failed with message: 2 UNKNOWN: library is not valid: missing header file "SparkFun_VL53L1X_Arduino_Library-master.h"

santaimpersonator, 2021-09-01:
Unfortunately, we don't provide support for the test (alpha or beta) versions of the Arduino IDE.

I am labeling this as won't fix since it has to do with a beta release of the Arduino IDE. I am also closing this issue; however, feel free to reopen the issue if it is still a problem when the 2.x.x version release.

closed 2021-09-01 by santaimpersonator

nseidle, 2021-09-01:
Just out of curiosity, I tested this with the latest beta and the lib installed without problems. The beta of v2 is still pretty rough but they are getting the edges rounded off. If anyone else still has an issue please let us know. We'll keep an eye on our libs WRT v2 as the beta gets solidified.

[image]