sparkfun/SparkFun_VL53L1X_Arduino_Library #47: API calls missing for implementing Distance based interrupts

chipmc, 2022-01-10:
The Sparkfun Library is missing two functions that that are part of the ST API.

https://www.st.com/resource/en/datasheet/vl53l1x.pdf

In the ST user manual on page 13, these are the parameters to set for range threshold:
Example:
Detectionconfig.DetectionMode = 1
Detectionconfig.Distance.CrossMode = 3
Detectionconfig.IntrNoTarget = 0
Detectionconfig.Distance.High = 1000
Detectionconfig.Distance.Low = 100
Status = VL53L1_SetThresholdConfig(&VL53L1Dev, &detectionConfig );
This function is used to program the device to report ranging only when an object is detected within 10 cm and 1 m (as in this example).
The function VL53L1_GetThresholdConfig() allows the programmed report threshold configuration to be obatined.

There are two items in this configuration that are missing in the SparkFun library : IntrNoTarget and DetectionMode

Without these two, there are a large number of "false" interrupts where the interrupt is raised but there is not a measurement in the designed range.

Adding these two API calls is not straight forward as ST does not expose the registers.

Would it be possible to add these two so a range limited interrupt implementation is possible?

PPVJM, 2022-01-21:
The functions were added as requested. Please pull the library again and check Example 8. Cheers!

closed 2022-01-21 by PPVJM

chipmc, 2022-01-21:
PPVJM - Thank you!

Any chance you could update in the Particle Library System.  It shows Sparkfun as the owner / maintainer.  

Thanks, Chip