sparkfun/SparkFun_VL53L1X_Arduino_Library #60: I2C address can't change

jendy12345, 2023-06-26:
Hello, I have purchased the product with part number 100080101-01-EVAL, which is listed as part number 2190-100080101-01-EVAL-ND. I encountered an issue where I need to change the sensor address. For example, I want to change the original sensor address from 18 to 30. Initially, I was able to successfully update the address, but after connecting via Bluetooth and powering off, the previously set sensor address reverted back to the default value and I am unable to change the address anymore. What could be the cause of this issue? How should I proceed if I still need to change the address?

I have encountered the same issue with both the Bluetooth module on the Bluno Beetle v1.1 and the HC-06 Bluetooth module connected to the Arduino UNO board. After modifying the address, it reverts back to the default address and cannot be changed again. Does this sensor support address modification? Currently, I have two devices with the address set to 18, but I need to modify three different addresses: 18, 30, and 40. How should I proceed to handle this?

I'm using this code to change the address. However, when I try to modify it, it doesn't change successfully in one attempt. I need to repeatedly refresh and modify it to make the change successful.I used the Example6_SetI2CAddress code file provided to make the modification.

[image]

Another issue I encountered is that I can successfully change the address when I set it to 18, but any other address setting fails.
It is possible that the sensor's address modification is locked or restricted？？

[image]

jendy12345, 2023-06-27:
@nseidle 
Hello, I have purchased the product with part number 100080101-01-EVAL, which is listed as part number 2190-100080101-01-EVAL-ND. I encountered an issue where I need to change the sensor address. For example, I want to change the original sensor address from 18 to 30. Initially, I was able to successfully update the address, but after connecting via Bluetooth and powering off, the previously set sensor address reverted back to the default value and I am unable to change the address anymore. What could be the cause of this issue? How should I proceed if I still need to change the address?

I have encountered the same issue with both the Bluetooth module on the Bluno Beetle v1.1 and the HC-06 Bluetooth module connected to the Arduino UNO board. After modifying the address, it reverts back to the default address and cannot be changed again. Does this sensor support address modification? Currently, I have two devices with the address set to 18, but I need to modify three different addresses: 18, 30, and 40. How should I proceed to handle this?

I'm using this code to change the address. However, when I try to modify it, it doesn't change successfully in one attempt. I need to repeatedly refresh and modify it to make the change successful.I used the Example6_SetI2CAddress code file provided to make the modification.
[image]

Another issue I encountered is that I can successfully change the address when I set it to 18, but any other address setting fails.
It is possible that the sensor's address modification is locked or restricted？？

[image]

Shamshala, 2023-06-27:
Why did repost the whole post? Do you use mere example code from this library or just a part of it? Address reverting after power off is normal.

Do you use XSHUT pin?

Shamshala, 2023-06-27:
Why did repost the whole thing? Do you use mere example code from this library or just a part of it? Address reverting after power off is by design, btw.

Do you use XSHUT pin?

Shamshala, 2023-06-27:
Why did you repost the whole thing? Do you use mere example code from this library or just a part of it? Address reverting after power off is by design, btw.

Do you use XSHUT pin?

jendy12345, 2023-06-27:
@Shamshala Sorry, because I don't know how to delete the previous post, because I suddenly found that I typed less.
What is XSHUT pin??
I use arduino UNO board for testing.

Shamshala, 2023-06-27:
Ok, sice you don't know XSHUT pin, that's probably your issue. The address change is per datasheet possible during sensor initialization ("boot"?) so if your code misses the right point after you power the whole gizzmo up, you are out of luck to do it afterwards. You better turn your gizzmo up while XSHUT set as output low, then flip it to an input and initialize the sensor during which change the address.

jendy12345, 2023-06-28:
But there is no XSHUT pin on my module, and my module only has RST, INT, SDA, SCL, GND, VCC pins on my module.

Shamshala, 2023-06-28:
Alright, then try to use the RST (assuming it's abbreviation for reset.

Shamshala, 2023-06-30:
So? Did you managed the code? Does it work?

Shamshala, 2023-06-30:
So? Did you manage the code? Does it work?

robomaniac, 2024-05-10:
here is snippet to change the address. I use pololu 

`#define INTERRUPT_PIN 12
#define SHUTDOWN_PIN 13

SFEVL53L1X distanceSensor(Wire, SHUTDOWN_PIN, INTERRUPT_PIN);

void setup(void)
{
  Wire.begin();

  Serial.begin(115200);
  

  Serial.println("VL53L1X Qwiic Test");
  
  distanceSensor.sensorOff();
  distanceSensor.setI2CAddress(0x52);
  distanceSensor.sensorOn();
  `

nsmith1024, 2024-08-04:
its amazing they dont show this in any of the example code, maybe they are just trying to frustrate us and waste a lot of our time chasing ghosts.